dalance/svlint #1: Suggestion for legacy always not to force always_ff

asicguy, 2019-10-29:
In FPGA designs you can use an initial block to set the startup value for a FF. However, VCS complains if you use an always_ff on the same signal that it's multiply driven. ex:

initial begin
  out_sig = '0;
end
always_ff @(posedge clk) out_sig <= in_sig;

would produce a multiply driven net error.

asicguy, 2019-10-30:
It would be nice if it could enforce using always_comb rather than always @* always @() etc... but not force the use of always_ff. It's certainly not a deal killer for me, it's more a limitation of SV that maybe they'll address in the future.
Thanks,
Frank

dalance, 2019-10-31:
For example, I can add a rule "level_sensitive_always" which check whether 'always' has edge identifier.

```verilog
module A;
always @* begin
end
always @ ( a or b ) begin
end
always @ ( posedge a ) begin
end
endmodule
```

```
Fail: level sensitive always
   --> testcases/fail/level_sensitive_always.sv:2:1
  |
2 | always @* begin
  | ^^^^^^ hint: level sensitive 'always' must be 'always_comb'

Fail: level sensitive always
   --> testcases/fail/level_sensitive_always.sv:4:1
  |
4 | always @ ( a or b ) begin
  | ^^^^^^ hint: level sensitive 'always' must be 'always_comb'
```

asicguy, 2019-10-31:
That would be great! Thanks.

closed 2019-10-31 by dalance