dalance/svlint #5: Error when parsing UVM macros

jeosadn, 2019-11-02:
I'm not sure whether I should be filing this against svlint or svparse, but since I'm the issue I'm having is hit when using svlint, I decided to file it here.

I have a minimal testbench that, when using UVM macros, hits the following parse error:
```bash
$ svlint --config .svlint.tom -f sample.vc -i ../1800.2-2017-1.0/src
Error: parse error
   --> ../1800.2-2017-1.0/src/macros/uvm_object_defines.svh:296:5
    |
296 |   `m_uvm_object_registry_internal(T,T)  \
    |     ^

```

You can reproduce this locally by downloading the attached files and calling:

```
make UVM_HOME=<your local UVM install>
```
[repro.tar.gz](https://github.com/dalance/svlint/files/3800163/repro.tar.gz)

dalance, 2019-11-03:
Thanks! I could reproduce it. I'll fix later.

dalance, 2019-11-05:
I fixed it and released v0.2.4.

commit 8c03c616 (2019-11-05): Update sv-parser to fix #5
closed 2019-11-08 by dalance