dalance/svlint #6: Support use of environment variables in dot-f format?

kevbroch, 2019-12-10:
My `design.f` file has something like `+incdir+$DSNTOP/inc` 

So I kept getting the following message until I just replaced $DSNTOP with the full path.
`Error: failed to include 'dsn_param.svh'`

Here's Icarus [Verilog doc description](https://iverilog.fandom.com/wiki/Command_File_Format?action=edit&section=4) if that's useful.

dalance, 2019-12-11:
Thank you for the request.
I added the support of variable with paren like Icarus at v0.2.19.

Now `+incdir+$(DSNTOP)/inc` or `+incdir+${DSNTOP}/inc` can be used.
I think `+incdir+$DSNTOP/inc` may be ambiguous and expect to use paren or brace.

kevbroch, 2019-12-12:
Thank you for this feature and how quickly you did it.

closed 2019-12-12 by kevbroch