dalance/svlint #8: Making dot-f parser into its own crate

Raamakrishnan, 2020-01-18:
Hi

Thank you for this wonderful set of projects. Your project made me to learn rust, just because I wanted to contribute to it.

I think it would be helpful if the dot-f file parser is made into its own crate. This crate can be used by svls (to provide better workspace-level features) and any other future projects that make use of sv-parser or svlint.

dalance, 2020-01-20:
Thank you for your proposal.
I'll try to split `parse_filelist` to new crate.

BTW, do you know the official name/definition of dot-f file?
(dot-f may be misunderstood as FORTLAN file...)

Raamakrishnan, 2020-01-20:
I am not aware of any official name for it. They are one of those unnamed standard among the tools. The name "filelist" seems fine.

If it is ok for you, I can work on this new filelist crate. I will use your `parse_filelist` as the starting point.

dalance, 2020-01-21:
Of course, OK.
`parse_filelist` can be used freely.

Raamakrishnan, 2020-01-21:
Great! Thanks!

closed 2020-01-21 by Raamakrishnan