dalance/svlint #11: One-line flag doesn't work with errors

sgregk, 2020-01-27:
Hi,

Your linter/parser are the only tools currently out there that can parse verilog/systemverilog files without doing full hierarchy elaboration, parsing includes, modules, parameter value etc. This makes it very suitable for a real-time syntax parser/linter, to be used with text editors such as Sublime, emacs, or in my case - vim. I'm currently writing a vim parser file for ALE (https://github.com/dense-analysis/ale), and with the latest version of svlint (0.3.1) - it works! Thank you for this tool - its really fast and looks like it supports every language construct.

One inconvenience in the current version of svlint is that the "-1" flag only applies to warnings. These are reported as

> Fail    testcase10.v:47:9       input wire               CLK,   hint: `input` must have `var`

Errors, on the other hand, are never condensed into one line:

> Error: parse error
>    --> testcase10.v:10:5
>    |
> 10 |     delay_line_10   
>    |     ^

Currently, I'm passing the above outcome through sed, and doing some regex magic to condense it into one line. Is is possible to make the "-1" argument apply to errors as well? The expected output should be something like:

> Error    testcase10.v:10:5       delay_line_10,   hint: parse error: error description if exists

This will make using the tool's output as a syntax/linter so much easier

Thank you,
Greg

dalance, 2020-01-28:
Thank you for your report.
I fixed it, and release v0.3.2.

sgregk, 2020-01-29:
Thank you very much - I've tested and verified it works as expected.

closed 2020-01-29 by sgregk