dalance/svlint #13: automatic subroutine check does not consider default lifetime

taichi-ishitani, 2020-02-19:
## Problem

The `default lifetime` can be specified in module/interface/package/etc, header definition.
However it's seemed that `automatic subroutine checker` does consider `default lifetime`.

## Expectation

```systemverilog
interface automatic test_if;
  function void test_func();
  endfunction
endinterface
```

`function_with_automatic` checker should not be failed because `test_func` is automatic function.

## Actual

`function_with_automatic` checker reports following error.

```
Fail: function_with_automatic
   --> test.sv:2:3
  |
2 |   function void test_func();
  |   ^^^^^^^^ hint  : `function` must be `automatic`
  |            reason: this causes mismatch between simulaton and synthesis
```

taichi-ishitani, 2020-02-19:
In addition, default lifetime of subroutines defined in class is automatic.
Therefore, `function_with_automatic ` checker should report no errors for following code:

```systemverilog
package test_pkg;
  class test_class;
    function void test_func();
    endfunction
  endclass
endpackage
```

However, the checker reports error below:

```
Fail: function_with_automatic
   --> test.sv:8:5
  |
8 |     function void test_func();
  |     ^^^^^^^^ hint  : `function` must be `automatic`
  |              reason: this causes mismatch between simulaton and synthesis
```

This checker is to avoid mismatch between simulation and synthesis so I think it might be better that the checker ignores subroutines defined in class.

dalance, 2020-02-19:
I fixed it and released. v0.4.1.

taichi-ishitani, 2020-02-19:
Thank you !
I've confirmed this issue is resolved.

closed 2020-02-19 by taichi-ishitani