dalance/svlint #15: Reports errors when file list includs empty lines

taichi-ishitani, 2020-02-19:
The svlint reports errors when input file list includes empty lines.
You can reproduce this issue by using the attached sample code.
[svlint.zip](https://github.com/dalance/svlint/files/4225626/svlint.zip)

Unzip the attached zip file and invoke command below. 

    $ svlint -f list.f -v

Then you will see the error message below.

```
$ svlint -f list.f -v
Info: pass 'foo.sv'
Error: File error: ""
Info: pass 'bar.sv'
Error: File error: ""
```

I think the tool should be ignore empty lines.

dalance, 2020-02-20:
Thanks.
I sent a patch to the dependent library.

dalance, 2020-02-21:
The issue was fixed, and I released v0.4.4.

taichi-ishitani, 2020-02-21:
I've confirmed this issue is fixed.
Thank you for your support !

closed 2020-02-21 by taichi-ishitani