dalance/svlint #18: Unexpected parse error on casting

taichi-ishitani, 2020-03-30:
I got an unexpected parse error on width casting.
You can re-produce the same error by using following code.

```systemverilog
package test_pkg;
  typedef struct packed {
    int width;
  } test_struct;
endpackage

module test_module
  import  test_pkg::*;
#(
  parameter test_struct TEST_PARAM  = '0
);
  logic [TEST_PARAM.width-1:0]  test;

  always_comb begin
    test  = TEST_PARAM.width'(0);
  end
/*
  localparam  int WIDTH = TEST_PARAM.width;
  always_comb begin
    test  = WIDTH'(0);
  end
*/
endmodule
```

```console
$ svlint  test.sv 
Config file '.svlint.toml' is not found. Enable all rules
Error: parse error
   --> test.sv:14:3
   |
14 |   always_comb begin
   |   ^
```

I changed `TEST_PARAM.WIDTH` to `WIDTH` then I saw no error.

I think `TEST_PARAM.width'(0)` should be accepted by following rules.

* cast ::= casting_type ' ( expression )
* casting_type ::= constant_primary
* constant_primary ::= ps_parameter_identifier constant_select
* constant_select ::= [ { . member_identifier constant_bit_select } . member_identifier ]constant_bit_select[ [ constant_part_select_range ] ]

dalance, 2020-04-01:
Thanks!
I fixed sv-parser library, and will release svlint later.

dalance, 2020-04-03:
I released v0.4.7.

taichi-ishitani, 2020-04-03:
Thank you!
I confirmed this error is resolved.

closed 2020-04-03 by taichi-ishitani