dalance/svlint #19: Timescale with `define macros

sgherbst, 2020-04-07:
Thanks for the awesome work on this project and sv-parser!  Just ran across a small issue when using `` `timescale `` in combination with `` `define `` macros.  I'm not sure about what the LRM says about this syntax, but it seems to be supported by some commercial and open-source simulators.

Example:
```verilog
`define TIME_UNIT 1ns
`define TIME_PRECISION 1ps
`timescale `TIME_UNIT / `TIME_PRECISION
module test;
    initial begin
        #(1.234321ns);
        $display($realtime);
        $finish;
    end
endmodule
```

Output:
```shell
> svlint test.sv
Config file '.svlint.toml' is not found. Enable all rules
Error: parse error
   --> test.sv:3:1
  |
3 | `timescale `TIME_UNIT / `TIME_PRECISION
  | ^
```

dalance, 2020-04-07:
Thank you for your report.
LRM doesn't say this syntax is valid or invalid.
So I'll fix this parse error.

DaveMcEwan, 2022-02-04:
This issue is already raised on [sv-parser](https://github.com/dalance/sv-parser) instead of svlint.
@dalance Would you close this issue?

closed 2022-02-07 by dalance