dalance/svlint #75: Per File and Per Line Lint Rule Disable

AlexLao512, 2021-05-26:
We are running into some issues where we have a few cases where we want to violate the lint rules we enabled but don't want to disable the lint rule for the whole file. Linting tools for other languages have a way to disable lint rules on an individual basis for a whole file or on a per line/per statement basis.

Something similar to how PyLint does things
`# pylint: disable=line-too-long,multiple-statements`

I think something like the following would work for SVLint.

`// svlint: file_disable=non_blocking_assignment_in_always_comb

if (value) a <= b; // svlint: disable=if_with_begin
`

DaveMcEwan, 2022-02-04:
@AlexLao512 Do you have any comments for PR #114?

AlexLao512, 2022-02-07:
> @AlexLao512 Do you have any comments for PR #114?

Hey, I wish GitHub sent me an email when I am tagged, I will have a look at what you have done. Thanks so much for looking into this!

DaveMcEwan, 2022-03-24:
@AlexLao512 or @dalance PR #114 has now been merged, so I think this can be closed.

closed 2022-03-24 by dalance

AlexLao512, 2022-03-24:
@DaveMcEwan Thanks so much for adding this!