dalance/svlint #77: Unexpected include error

taichi-ishitani, 2021-06-03:
I got an unexpected include error when linting SV file which has a macro call which of which an argument is another macro like below.

```verilog
`define PATH_BAR  bar.sv
`define QUATE(path) `"path`"
`include  `QUATE(`PATH_BAR)
```

[unexpected_include_error.tar.gz](https://github.com/dalance/svlint/files/6592332/unexpected_include_error.tar.gz)

I attached a sample test case. You can use reproduce this error by using this:

```
$ tar -xvf unexpected_include_error.tar.gz
$ cd unexpected_include_error
$ svlint foo.sv
```

Then you will see the error message below.

```
Config file '.svlint.toml' is not found. Enable all rules
Error: failed to include '`PATH_BAR'
```

taichi-ishitani, 2021-06-03:
Following case will also raise a similar error.

```verilog
`define PATH_BAR bar.sv
`define QUATED_PATH_BAR `"PATH_BAR`"
`include `QUATED_PATH_BAR 
```

```
$ svlint foo.sv
Config file '.svlint.toml' is not found. Enable all rules
Error: failed to include '`PATH_BAR'
```

DaveMcEwan, 2022-02-04:
This issue should be raised on [sv-parser](https://github.com/dalance/sv-parser) instead of svlint.
@dalance Would you close this issue?

closed 2022-02-07 by dalance