dalance/svlint #99: `input must have var` clarification

Timmmm, 2021-11-02:
If you follow the (good IMO) svlint recommendation to use `default nettype none` and then add the required `var`s to module ports you might end up with code like this:

```
`default_nettype none

module foo (
  input var logic i_a,
  output var logic o_b
);
```

svlint is happy with that. However the default net type for `input` ports is actually `wire`, not `var`. If you put `input wire` (or `input tri`) then svlint says you should use `input var`.

I'm not an SV pro - is there actually any difference between `wire` and `var` here, and what's wrong with using `wire`?

dalance, 2021-11-12:
We want to use `logic` and use `default nettype none`.

```systemverilog
input x            // NG: net_type is none
input wire x       // OK
input logic x      // NG: net_type is none
input var logic x  // OK
input var x        // OK

output x           // NG: net_type is none
output wire x      // NG: can't use x in always_ff
output logic x     // OK
output var logic x // OK
output var x       // OK
```

So we think the simplest rule is `input var / output var`,
but `input wire / output var` is no problem.
If you want to use it, new rule can be added.

Timmmm, 2021-11-12:
I definitely agree. I finally found the bit of the LRM that describes the difference between wire and var, and am I right in thinking that the main difference is that you can have multiple drivers for a wire? Which is a mistake 99% of the time (buses are the only exception I can think of), and that's why `input var` is ok - probably better even than `input wire`?

Timmmm, 2021-11-12:
Btw maybe worth adding that motivation to the rules page.

dalance, 2021-11-12:
> I definitely agree. I finally found the bit of the LRM that describes the difference between wire and var, and am I right in thinking that the main difference is that you can have multiple drivers [...]

Surely, `var` seems to be better than `wire`.
In our rule, we use `inout tri` for bidirectional bus.

```systemverilog
module A(
    input wire a,
    input var  b,
    inout tri  c
);

    assign a = 0; // Compile pass, but wrong assignment
    assign b = 0; // Compile error
    assign c = 0; // OK
endmodule
```

dalance, 2021-11-12:
I added the reason of the `input var` rule.

https://github.com/dalance/svlint/blob/master/RULES.md#input_with_var

Timmmm, 2021-11-12:
Ah I think I understand it all now, thanks!

closed 2021-11-12 by Timmmm