dalance/svlint #158: Some Rules Cause TOML Compilation to Fail

cvanbeek13, 2022-04-05:
I'm working on setting up a default `.svlint.toml` file for my company.  There's a few rules that we wanted to disable for our own coding standards, so it appears that the only option is to create the toml file and specify all the rules we want to turn on using `<rule> = true`.

Here is my current toml file:

```toml
[rules]
blocking_assignment_in_always_ff = true
case_default = true
default_nettype_none = true
enum_with_type = true
# explicit_case_default = true
# explicit_if_else = true
for_with_begin = true
function_same_as_system_function = true
function_with_automatic = true
generate_case_with_label = true
generate_for_with_label = true
generate_if_with_label = true
generate_keyword_forbidden = true
generate_keyword_required = true
genvar_declaration_in_loop = true
genvar_declaration_out_loop = true
if_with_begin = true
inout_with_tri = true
input_with_var = true
interface_port_with_modport = true
legacy_always = true
level_sensitive_always = true
# localparam_explicit_type = true
# localparam_type_twostate = true
loop_variable_declaration = true
lowercamelcase_interface = true
lowercamelcase_module = false
lowercamelcase_package = false
non_ansi_module = true
non_blocking_assignment_in_always_comb = true
output_with_var = true
# parameter_explicit_type = true
parameter_in_package = true
# parameter_type_twostate = true
prefix_module = true
prefix_package = false
priority_keyword = true
# sequential_block_in_always_comb = true
# sequential_block_in_always_ff = false
# sequential_block_in_always_latch = false
tab_character = true
unique0_keyword = true
unique_keyword = true
uppercamelcase_package = true
uppercamelcase_module = false
wire_reg = true
```

All of the rules which I've commented out (`#`) cause the compilation to fail if I include them.  It doesn't matter if they're true or false.  I get the following error: `Failed to parse d:\<path>\.svlint.toml. Enable all lint rules.`.  Is there anything I'm doing wrong for these rules, or is this a bug?

I am using the [VSCode plugin](https://github.com/dalance/svls-vscode).

cvanbeek13, 2022-04-05:
I think you're right, I just downloaded the 0.5.1 release and am running the `svls.exe` file in there.

What are the steps involved to build it from source?  That would allow me to add my own custom rules directly as well, which would be a plus.  Do I just need to [install cargo and rust](https://doc.rust-lang.org/cargo/getting-started/installation.html) and run the Makefile?

DaveMcEwan, 2022-04-06:
> Do I just need to [install cargo and rust](https://doc.rust-lang.org/cargo/getting-started/installation.html) and run the Makefile?

Yes :) @dalance has made it nice and easy. You can build svlint as a standalone command-line utility. Alternatively, you build svls and that will include svlint.

cvanbeek13, 2022-04-06:
Awesome, that worked. I was able to rebuild svls and now it's working properly with the vscode plugin.  Thanks @DaveMcEwan !

dalance, 2022-04-07:
I released svlint v0.5.2 and svls v0.2.1.

DaveMcEwan, 2022-04-20:
@cvanbeek13 Would you close this if you think it's resolved, please.