dalance/svlint #176: New Rule: Parameter Must Have Default Value

AlexLao512, 2022-06-22:
We have been having some really weird issues with Vivado where its source scanner (srcscanner.exe or srcscanner on Linux) process will crash with no log file and return "[filemgmt 56-315] Source scanning failed during design analysis. To get more details run synthesis or simulation and check the log.". This happens in two different somewhat recent versions of Vivado. Running synthesis will report that some files are missing. Looking more closely at the Sources -> Hierarchy tab of Vivado we noticed some of our modules have instances names that match some of our parameters instead of modules, we concluded that somehow maybe the srcscanner's internal data structures were being corrupted, after many days changing a lot of stuff across our code base to figure out what was going on we have a handful of rule suggestions to make. 

One of the things we did that seemed to help the problem was give every module and interface parameter a default value. 

Example
```
ALLOWED
module example_module #(
    parameter int CLOCK_FREQ = 1000
) (
    input clk
);

NOT ALLOWED
module example_module #(
    parameter int CLOCK_FREQ
) (
    input clk
);
```

DaveMcEwan, 2022-06-29:
I like this idea, and it's relatively simple to implement.
Just need to check that the 3rd element of every `ParamAssignment.nodes` is `Some(_)` rather than `None`, but only within a ParameterPortList.
https://docs.rs/sv-parser/latest/sv_parser/struct.ParamAssignment.html
https://docs.rs/sv-parser/latest/sv_parser/enum.ParameterPortList.html

DaveMcEwan, 2022-11-09:
@AlexLao512 Would you have a look at the master branch, and close this if you're happy?
Have you any updates on the srcscanner issue? It's an interesting issue, for sure.

AlexLao512, 2022-11-09:
@DaveMcEwan The changes on master look good, I will build from sources and test at work when I get a chance. 

The issues with Vivado seem to be solved when we give every parameter a default value, it does occasionally come back when we forget so this rule will be very handy to have.

AlexLao512, 2022-11-09:
@DaveMcEwan I ran a source build with this rule enabled on a 1/4 million line code base. Looks good, it found a few instances of parameters without default values we missed.

DaveMcEwan, 2022-11-16:
Good to know it's doing useful work :)
Would you close the issue if you're satisfied?

closed 2022-11-16 by AlexLao512