dalance/svlint #177: Question regarding sequential_block_in_always_ff

Masshat, 2022-06-29:
Hello, I tarted using svlint in order to improve the quality of the RTL I write and keep all my code consistent overall, I also use svlint to check if there is any syntax error, I find that faster then calling a commercial EDA compiler/simulator.

I'm the kind of guy who uses begin/end inside my always_ff construct, however svlint seems to not like it, I know I can disable the check but I'm curious to know what kind of problem can occur while using begin/end within always_ff construct. 
The tool hint log state the following:

>  hint  : begin/end forbidden within `always_ff` constuct
    reason: prevent introducing sequential dependencies.

Can someone explain me what does it mean ?

Thanks,
Mass.

DaveMcEwan, 2022-06-29:
Hi @Masshat,
This is actually my favourite rule, though the consequences/purposes are a bit subtle. I recommend this rule particularly to students and new designers to ensure that they can always draw an accurate logical diagram of what their code infers.

With a sequential block (marked by `begin` and `end`) you can assign to multiple signals in a leaf conditon which can easily result in difficult-to-comprehend logic, e.g.:
```systemverilog
always_ff @(posedge clk) begin
  if (cond) begin
    foo_q <= foo_d;       // Block was originally written for foo.
    bar_q <= bar_d;       // This was added later.
  end
  bar_q <= bar_d;         // What happens to bar_q?
end
```
Without a sequential block, you enforce that only one signal is assigned to per leaf condition.
A nice consequence is that you *know* that only one signal is updated each time the `always_ff` block is evaluated.
IEEE1800-2017 specifies that if a signal is assigned to in an `always_ff` block, then it shall not be assigned to by any other block (compile error, Clause 9.2.2.4) - a key axiom for the purpose of this rule.

An example with multiple signals in the `always_ff` is a ping-pong buffer (AKA shunt buffer, buffer of a 2-entry fifo).
Due to the construction, you can be sure that you never update both entries at the same time (except on reset).
```systemverilog
  // Enforced exclusive updates, with reset and clockgate.
  always_ff @(posedge clk)
    if (rst) {ping_q, pong_q} <= '0;
    else if (clkgate)
      if (foo) ping_q <= data_q;
      else     pong_q <= data_q;
  // Optional explicit else here:
  //else
  //  {ping_q, pong_q} <= {ping_q, pong_q};
```

Another example with multiple signals is an address decoder.
Due to the construction, you can be sure that you aren't accidentally updating multiple registers on a write to one address.
```systemverilog
  // Enforced exclusivity of address decode.
  always_ff @(posedge clk)
    if (write)
      case (addr)
        123: red_q   <= data_q;
        456: blue_q  <= data_q;
        789: green_q <= data_q;
      endcase
```

When you don't need those exclusivity properties, only one signal should be updated per `always_ff`.
That ensures that the code doesn't get too deep/complex/unintuitive, and drawing a logical diagram is straightforward.
I'd expect this to be the style for most signals.
```systemverilog
  always_ff @(posedge clk)
    if (rst)          ctrl_q <= '0;
    else if (clkgate) ctrl_q <= ctrl_d;
    // Optional explicit else here:
    //else            ctrl_q <= ctrl_q;
```

DaveMcEwan, 2022-06-29:
FYI, I'm working (slowly) on PR #170 to add explanations for each rule, but it's taking a while with other life/work getting in the way :p

AlexLao512, 2022-06-29:
I would like to note here that many places that write RTL ALWAYS require begin/end contrary to the rule here. Excluding begin/end can be very error prone when editing code after the fact. 

There is a longer list of reasoning and more strict styling for surrounding we follow that requires begin/end so we choose to disable this rule.

DaveMcEwan, 2022-06-30:
@AlexLao512 That's a good point about the surrounding environment of other rules, i.e conjunction of rules, AKA ruleset.

This rule works together with `explicit_case_default`, `explicit_if_else`, and `default_nettype_none` to give a useful set of properties. One of those properties is actually that editing code after the fact is becomes safe/not-error-prone...[full explanation missing :p]
Of course, as you say there are other ways to construct rulesets with that property.

I'm keen to see different ways of doing thing. Would you share your longer list of reasoning that I can incorporate into #170?

AlexLao512, 2022-06-30:
@DaveMcEwan I will have a look, I will note for @Masshat that use of any linter for any language always involves carefully selecting rules. 

There is a philosophy for individual rules to be put in #170 but we might want to put together a wiki page with a curated rule set and philosophy for the the entire set for beginner users to go through. I am putting together my own repo of an example HDL development repository with automation, unit testing tools/methods and styling. SVLint will be included. Unfortunately I haven't had time to work on it.

DaveMcEwan, 2022-06-30:
> There is a philosophy for individual rules to be put in #170 but we might want to put together a wiki page with a curated rule set and philosophy for the the entire set for beginner users to go through.

You read my mind :p My intention is to put as much detail and commentary into #170 including important interactions with other rules, and to update the README in [svlint-rulesets](https://github.com/DaveMcEwan/svlint-rulesets) to cover philosophies behind whole sets.
Maybe that's not the ideal places, but I'll worry about location once the (high effort) content is ready. Don't we all need more time!

DaveMcEwan, 2022-07-11:
@Masshat Does this conversation answer your question? If not, what is unanswered? Otherwise, would you close the issue please.

closed 2022-11-27 by Masshat