dalance/svlint #178: All class scoped variables should have a prefix of "m_"

abhishek-elastics, 2022-07-01:
All class member variables should begin with "m_". Conversely, if a variable begins with "m_", it should be a class scoped variable and not a method scoped variable

# Pass
```
class abc extends uvm_sequence_item;
   int m_foo;
   string m_bar;

   function void dont_care();
     int foo;
     int bar;
   endfunction: dont_care
endclass: abc
```

# Fail
```
class abc extends uvm_sequence_item;
  int m_foo;
  string bar;                               // Class scoped variable doesn't begin with "m_"
  
  function void dont_care();
    int m_foo;                              // Non-class (local) variable begins with "m_"
    int bar;
  endfunction: dont_care
endclass: abc
```

DaveMcEwan, 2022-11-04:
@abhishek-elastics I think this could be addressed by adding some rules to #195.
I'm thinking of these 4 rules: `re_(forbidden|required)_(class|function)_var`
Is this something you'd like to add to my PR?

abhishek-elastics, 2022-11-04:
Hey @DaveMcEwan That would be great. method could be a ```task``` as well. Although I'm not sure by what you meant by "add"? Do you want me to commit to your PR?

DaveMcEwan, 2022-11-06:
I've just had a go at implementing them myself on a derived branch `reRules-dev`.
Would you like to test it to see if it fits your needs?
If you're satisfied, then I can easily merge it into #195.

1. Clone and build my dev branch.
```
git clone git@github.com:DaveMcEwan/svlint.git svlint-TEST178
cd svlint-TEST178
git checkout reRules-dev
cargo build
```

2. Configure the rexeges in a `test178.toml` file with something like this:
```toml
[option]
#re_forbidden_var_class = ""
re_forbidden_var_classmethod = "^m_"
re_required_var_class = "^m_[a-z]+[a-z0-9]*$"
#re_required_var_classmethod = ""

[rules]
#re_forbidden_var_class = true
re_forbidden_var_classmethod = true
re_required_var_class = true
#re_required_var_classmethod = true
```

3. Run svlint on some of your files like this:
```
SVLINT_CONFIG=test178.toml cargo run --bin=svlint -- path/to/file.sv
```

abhishek-elastics, 2022-11-08:
@DaveMcEwan I ran the pass and fail examples from above and they worked as expected. These were the errors generated for the fail case
```python
Fail: re_required_var_class
   --> fail.sv:3:10
  |
3 |   string bar;                               // Class scoped variable doesn't begin with "m_"
  |          ^^^ hint  : Use a class-scoped variable identifier matching regex "^m_[a-z]+[a-z0-9]*$".
  |              reason: Identifiers must conform to the naming scheme.

Fail: re_forbidden_var_classmethod
   --> fail.sv:6:9
  |
6 |     int m_foo;                              // Non-class (local) variable begins with "m_"
  |         ^^^^^ hint  : Use a method-scoped variable identifier matching regex "^m_".
  |               reason: Identifiers must conform to the naming scheme.
```
For ```re_forbidden_var_classmethod```, it'd be better if the failure message read something like this instead ```Do not use a class-scope variable identifier in a method. matching regex "^[^m_]"```

DaveMcEwan, 2022-11-08:
Thanks @abhishek-elastics. Unfortunately, I'm not sure there's a clean way to make a custom hint/reason for the regex rules (because the regexes are user-configurable).
If its any consolation, I've already fixed the typo (missing "not") in the forbidden rule.

DaveMcEwan, 2022-12-18:
@abhishek-elastics The rules are now available in the latest release (v0.6.1). If you are satisfied, would you close this issue?

closed 2023-01-17 by abhishek-elastics