dalance/svlint #180: misspelling "tab_charactor"

TakaakiHARADA, 2022-07-12:
Hi
Just a tiny problem. The sample configuration generated by `svlint --example` has a misspelling -- "tab_charactor".
Supposed to be "tab_character"?
[image]

DaveMcEwan, 2022-07-12:
@puppuccino This has already been addressed long ago. The last 3 rules in your screenshot (including "tab_charactor") are aliases for renamed rules, which are kept for backwards compatibility.
See the source here: https://github.com/dalance/svlint/blob/master/build.rs#L14
Would you close the issue please.

puppuccino, 2022-07-12:
@DaveMcEwan 
I didn't notice that. Thank you!

closed 2022-07-12 by puppuccino

dalance, 2022-07-13:
This behaviour may be confusing. So I removed renamed rules from example.