dalance/svlint #182: Rule `explicit_if_else` only warns the first time in each `always_*` block

supleed2, 2022-07-28:
The same behaviour was observed with both
- `svlint "0.5.5 ( rev: 0aa25c3, rustc: 1.62.0, build at: 2022/07/06 00:34:49 )"`
- `svls "0.2.4 ( rev: 9586d6c, rustc: 1.62.0, build at: 2022/07/06 03:26:29 )"`

I have made a few test cases to show what I mean:
```systemverilog
module test_fine1;
  logic ck, a, b, c, d;
  always_ff @(ck)
    if (a) // GOOD: warns here
      b <= c;
  always_ff @(ck)
    if (b) // GOOD: warns here
      c <= d;
endmodule

module test_fine2;
  logic ck, a, b, c, d;
  always_ff @(ck) begin
    if (a)
      b <= c;
    else
      b <= d;
  end
  always_ff @(ck)
    if (b) // GOOD: warns here
      c <= d;
endmodule

module test_notfine1;
  logic ck, a, b, c, d;
  always_ff @(ck) begin
    if (a) // GOOD: warns here
      b <= c;
    if (b) // BAD: DOES NOT warn here
      c <= d;
  end
endmodule

module test_notfine2;
  logic ck, a, b, c, d;
  always_ff @(ck) begin
    if (a)
      b <= c;
    else
      b <= d;
    if (b)  // BAD: DOES NOT warn here
      c <= d;
  end
endmodule
```
I'm not sure if the issue of only applying once is important for / affects other rules.

DaveMcEwan, 2022-11-07:
@supleed2 A fix has been merged into master, and a new version released (v0.6.0). Would you care to check that this fixes your testcases and close this issue if it's okay?

supleed2, 2022-11-07:
Yep, seems to have fixed it 👌

closed 2022-11-07 by supleed2