dalance/svlint #192: Add option to force color output

prokie, 2022-10-28:
In some CI environments for example GitLab-CI with docker containers makes svlint think that it cannot use colored output, when in fact it is possible.

Is it possible to add this option for svlint?

DaveMcEwan, 2022-10-28:
Just to clarify for future readers: svlint will use colored output wherever the environment and context allows it.
Some contexts cannot use colored output, e.g. redirecting STDOUT to a file `svlint foo.sv > output.rpt`.
Similarly, some terminal environments don't support color, and there are different ways of encoding color information. The usual way of figuring out your terminal environment is to use `env` and look at the variables `TERM`, `COLORTERM`, `CLICOLOR`, and `CLICOLOR_FORCE`.

Further reading:
- https://bixense.com/clicolors/
- https://docs.rs/clicolors-control/latest/clicolors_control/

DaveMcEwan, 2022-12-15:
@prokie Would you mind closing this issue if it's no longer causing you problems?

DaveMcEwan, 2023-02-01:
@dalance I think this can be safely closed.

closed 2023-02-01 by prokie