dalance/svlint #210: BUG: Unable to disable default_nettype_none check per file

AlexLao512, 2022-11-23:
It seems like if I enable the default_nettype_none check I am unable to disable using a comment even if I put that comment at the top of the file.

DaveMcEwan, 2022-12-15:
I'm not sure this is a bug or a feature of the rule!
The rule is to check that the *first* thing in a `source_text` is the directive.
As the rule is only applied at exactly one point of a file, I can't think why you'd want to turn it off temporarily.
I agree that the following is confusing:
```systemverilog
/* svlint off default_nettype_none */
// The above control comment has no effect.
`default_nettype_none
module M;
endmodule
```
However, that's necessary by the rule's definition.

If you want to turn it of completely for a single file, I think that would be better addressed by using a different configuration.
```sh
# Some files should be checked with the rule enabled.
SVLINT_CONFIG="path/to/someRules.toml" svlint --filelist somefiles.fl

# Other files should be checked without the rule enabled.
SVLINT_CONFIG="path/to/otherRules.toml" svlint --filelist otherfiles.fl
```

If you have a tool which doesn't support the directive, this might be a suitable workaround.
```systemverilog
`ifdef DEFAULTNETTYPE_SUPPORTED
`default_nettype none
`endif
module M;
// ...
endmodule
```

An alternative rule could check for something like: "On encountering a `module`, check that `default_nettype` has been explicitly set to  `none`."
You could then do something like this:
```systemverilog
/* svlint off module_default_nettype_none */
module M;
endmodule
/* svlint on module_default_nettype_none */
```

AlexLao512, 2022-12-31:
The problem is some modules were written requiring default nettype wire and it is not worth our time refactoring now. Disabling outside of the file is harder to maintain and isn't as obvious to any future people reading the code.

DaveMcEwan, 2023-03-14:
@AlexLao512 Would you like to try the latest on the master branch and close this issue if #238 meets your needs?

AlexLao512, 2023-03-14:
Hey, thanks, I will have look sometime this week.

AlexLao512, 2023-03-14:
Looks good here.

DaveMcEwan, 2023-03-20:
@dalance Would you mind making a new release so this can be closed?

dalance, 2023-03-22:
I released v0.7.0.

DaveMcEwan, 2023-03-23:
Nice one, thanks :)
@AlexLao512 Can this be closed now?