dalance/svlint #228: cargo: compilation error

adrianf0, 2023-02-08:
Installing svlint in the  'ubuntu:20.04' docker container with cargo gives compilation error:
```
   Compiling sv-parser v0.12.3
error[E0599]: no variant or associated item named `DeferredImmediateAssetionItem` found for enum `RefNode` in the current scope
  --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/svlint-0.6.1/src/rules/re_forbidden_assert.rs:51:22
   |
51 |             RefNode::DeferredImmediateAssetionItem(x) => {
   |                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |                      |
   |                      variant or associated item not found in `RefNode<'_>`
   |                      help: there is a variant with a similar name: `DeferredImmediateAssertionItem`

error[E0599]: no variant or associated item named `DeferredImmediateAssetionItem` found for enum `RefNode` in the current scope
  --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/svlint-0.6.1/src/rules/re_required_assert.rs:51:22
   |
51 |             RefNode::DeferredImmediateAssetionItem(x) => {
   |                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   |                      |
   |                      variant or associated item not found in `RefNode<'_>`
   |                      help: there is a variant with a similar name: `DeferredImmediateAssertionItem`

For more information about this error, try `rustc --explain E0599`.
error: could not compile `svlint` due to 2 previous errors
warning: build failed, waiting for other jobs to finish...
error: failed to compile `svlint v0.6.1`, intermediate artifacts can be found at `/tmp/cargo-installIlXaTl`
``` 

Steps to reproduce:
```
$> docker run -it --rm ubuntu:20.04
$> apt update -y && apt install cargo
$> cargo install svlint
```

dalance, 2023-02-08:
Sorry.
sv-parser v0.12.3 has API breaking changes from v0.12.2.
I yanked v0.12.3, so could you retry?

adrianf0, 2023-02-08:
It work now, thanks!

closed 2023-02-08 by adrianf0