dalance/svlint #245: [feature] Shell completions

Freed-Wu, 2023-05-17:
Can this program provides completions for common shells? I guess <https://docs.rs/clap_complete/latest/clap_complete/> can be a helper. Same as <https://github.com/dalance/svls>

DaveMcEwan, 2023-05-17:
Both svlint and svls use clap in the same way to construct the command-line interface.
@Freed-Wu Can you provide some examples of what you'd like to see?

Freed-Wu, 2023-05-17:
Some examples: <https://github.com/typst/typst/issues/545> <https://github.com/git-lfs/git-lfs/issues/5104>

DaveMcEwan, 2023-05-17:
Sounds like shell completions could be useful but I'm still not exactly sure what you're after. Perhaps you could open a pull request?

Freed-Wu, 2023-05-17:
I am not familiar with rust :cry: I just think shell completions is useful so I hope every program I like can support it.

Freed-Wu, 2023-06-30:
Works for me!

PR #258 (2023-08-09): Improvement: Cleanup the command-line interface.
- Rename options to make the whole interface consistent and intuitive for new users.
  - `--example` -> `--config-example`
  - `--update` -> `--config-update`
  - Backwards compatibility maintained with aliases.
- Rename options to be consistent with other tools (GCC, Clang, Verilator, etc.).
  - `-i/--include` -> `-I/incdir` (also consistent with filelist plusarg/keyword)
  - `-d` -> `-D`
  - Backwards compatibility maintained with aliases.
- Remove unnecessary choice `--dump-filelist=no`.
  Technically a breaking change, but I doubt it is ever [... cut]
commit 51476e8e (2023-08-09): cli --dump-completion Print a shell completion to STDOUT.

- Fixes #245
- Based on example: <https://github.com/clap-rs/clap/blob/clap_complete-v3.2.5/clap_complete/examples/completion-derive.rs>
- `svlint --dump-completion=bash > /etc/bash_completion.d/svlint`
- <https://serverfault.com/questions/506612/standard-place-for-user-defined-bash-completion-d-scripts>
closed 2023-06-30 by Freed-Wu