dalance/svlint #252: New Rule: Check that filename matches identifier.

DaveMcEwan, 2023-06-07:
Modules, interfaces, packages, and programs are often requried to be declared in their own file.
A new set of rules should check, for example, that `module Foo; ... endmodule` lives in `some/path/to/Foo.sv`.

ShantanuPSinha, 2024-03-21:
Hello, I was trying to work on this issue but I ran into an issue with grabbing the file names. As far as I can find, none of the nodes seem to contain the filename information.

I was wondering if you have any pointers/know where I should look?

ShantanuPSinha, 2024-03-28:
```
let path = if let Some(x) = unwrap_locate!(node.clone()) {
    if let Some((path, _beg)) = syntax_tree.get_origin(&x) {
        Some(path)
    } else {
        None
    }
```

I got the file path from the Syntax tree, however, I am unable to get Cargo Test to pass the testcases.

As far as my understanding goes, Cargo Test modifies the input file name that is passed to the syntax rule checker. Printing out the file name:

```./svlint/target/debug/build/svlint-0484465ec2310bbe/out/syntaxrules.identifier_matches_filename.pass.1of1.sv```

For the pass testcase (in pass/identifier_matches_filename.sv), I have

```
module identifier_matches_filename;
endmodule
```

However, this fails, due to the editing of the filename prior to running by cargo test. Furthermore, it doesn't seem like I can edit the filename in any way for it to pass due to the periods (```.```) in the filename. That is to say, I cannot modify the pass testcase to be 

```
module syntaxrules.identifier_matches_filename.pass.1of1;
endmodule
```

due to this being an invalid module name.

I modified [build.rs (Line :380)](https://github.com/dalance/svlint/blob/0530ce6c5fef39ff0796a1fd3c0b31ff9856a672/build.rs#L380) to change the periods in the filename to underscores. This makes ```syntaxrules_identifier_matches_filename_pass_1of1``` a valid module name and cargo test now passes for all testcases.

This is one option. Alternatively, I could turn off that specific testcase, if there is a reason for wanting to keep the periods in the test suite filenames.

closed 2024-06-03 by dalance