dalance/svlint #262: Bug: parse error regarding `inside` and conditional generate

jscheid-ventana, 2023-07-21:
Using svlint-v0.8.0-x86_64-lnx.zip.

Testcase at top.sv
```
module top;
  typedef enum {
    APPLE,
    BANANA
  } fruit_t;

  parameter fruit_t FRUIT;

  if (FRUIT inside { APPLE }) begin
  end
endmodule
```

```
svlint --example > svlint.toml
svlint --config=svlint.toml top.sv
Error: parse error
   --> ../top.sv:9:13
  |
9 |   if (FRUIT inside { APPLE }) begin
  |             ^

```

Other tools we use, open and commercial, have no problem with this, and it seems to me that this is valid syntax.

jscheid-ventana, 2023-07-21:
Kudos, though, that this was the single parse error in a large set of design files.

DaveMcEwan, 2023-07-24:
@jscheid-ventana This is invalid syntax according to the BNF in IEEE1800-2017 Annex A. Essentially, you cannot depend on the result of `inside` being known at compile/elaboration time.

The relevant parts of the BNF are: 
```
# A.4.2 Generated instantiation, page 1159
if_generate_construct ::=
    if ( constant_expression ) generate_block [ else generate_block ]

# A.8.3 Expressions, page 1172
constant_expression ::=
    constant_primary
  | unary_operator { attribute_instance } constant_primary
  | constant_expression binary_operator { attribute_instance } constant_expression
  | constant_expression ? { attribute_instance } constant_expression : constant_expression
```

That is, the condition of a conditional generate must be `constant_expression` (different from `expression`), and there are only 4 variants of `constant_expression`, but none of them have a way to include an `inside_expression`.
The definitions of `expression` and `inside_expression` are on page 1173.

In this case, the other tools are being more forgiving than required.
[Sv-parser](https://github.com/dalance/sv-parser), (the parser underlying svlint) adheres very closely to the formal grammar, so I think this will be categorized as not-a-bug.

However, by adding an `always_comb` or `always @*` before the `if (FRUIT`, the syntax becomes valid. Maybe this can help your usecase?
I'm not 100% sure that all synthesis and formal-verif tools will treat it correctly, but simulations might be okay.

jscheid-ventana, 2023-07-24:
Baffling subsetting in the grammar.  I appreciate you digging further into the tree than I.  In our code base, I'll just be replacing the `inside` with an equivalent comparison+OR expression.

closed 2023-07-24 by jscheid-ventana