dalance/svlint #269: Feature request: Include a file before linting based on config setting

Sv3n, 2023-11-07:
Simulators and other tools can sometimes include certain files automatically. In particular, when files are written that use UVM classes, they often just assume that the appropriate uvm macros are available in the namespace. Just as a random example to illustrate the point:

https://github.com/chenyangbing/UVM-example/blob/master/agent.sv

The line `class mem_agent extends uvm_agent;` presumes that `uvm_agent` is available, and later it also relies on `uvm_component_utils`. In the two simulators that I've tested (Questasim, VCS), this construction actually works. When I give such a file to svlint, it accepts the class extension, but it errors out at `Error: macro 'uvm_component_utils' is not defined`. To fix this, I can add `include "uvm_macros.svh" at the top of the file (with the appropriate `--incdir` set). Then it works as expected, the file gets fully linted.

It would be nice to be able to mimic the behavior of the simulators with svlint, e.g. by adding an option to `.svlint.toml` to auto-include certain files (or more generally, execute preprocessor statements, although that is more than is required for what I had in mind here). Alternative could be a CLI option, although having it in the config file would make it automatically available for users of tools that use svlint indirectly (like https://github.com/dalance/svls).

DaveMcEwan, 2023-11-11:
I'm not sure that the config file is the best place but I like the general idea :) Here's my reasoning:

1. In a config shared between Linux, Mac, and Windows users, any paths will cause problems because Windows just does filepaths too differently! The option `exclude_paths` also suffers in this respect, but its use is quite limited and I think your feature would be used much more.
2. Absolute paths in a config file would lead to confusion when the config is shared between users and systems.
3. Relative paths in a config file are less of a concern, but I don't think the starting point is obvious. That is, do you take the starting point as the working directory, location of the config file, a git repository root, or maybe somewhere else?
4. The existing option `exclude_paths` doesn't have quite the same problem as what you're asking for, because it would usually refer to files in the same repo, whereas it sounds like you're asking to refer to something external (like a UVM directory used by several projects).
5. Having the paths in the config doesn't automatically lead to compatibility with svls. We would need a set of matching changes on both svlint and svls. Note, reporting an error in svls (about one of the auto files) might another usability issue, but let's come back to that later :p
6. I agree that a CLI flag would not be user-friendly! This feature should allow "setup and forget" but also allow full flexibility in order to work with Makefiles, scripts, etc.

So, how about 3 environment variables like `SVLINT_INCDIRS`, `SVLINT_PREFILES` and `SVLINT_POSTFILES`? I imagine them to look like the usual `PATH`, `MANPATH`, etc. with colons separating their lists of files, so they can be used with Environment Modules' [`prepend-path`](https://modules.readthedocs.io/en/latest/modulefile.html#mfcmd-prepend-path) and similar. It would also fit nicely into the ruleset wrapper scripts like `SVLINT_CONFIG` and work consistently between Windows and everything else.

For example:
```sh
$ export SVLINT_INCDIRS="/cad/tools/uvm:/work/myTeam"
$ export SVLINT_PREFILES="uvm_macros.svh:/another/thing/first"
$ export SVLINT_POSTFILES="/another/thing/last:check_pp_final_state.svh"

$ svlint foo.sv bar.sv

$ svlint hello.v world.v
```

Which would be exactly equivalent to:
```sh
svlint -I/cad/tools/uvm -I/work/myTeam \
  uvm_macros.svh \
  /another/thing/first \
  foo.sv bar.sv \
  /another/thing/last \
  check_pp_final_state.svh

svlint -I/cad/tools/uvm -I/work/myTeam \
  uvm_macros.svh \
  /another/thing/first \
  hello.v world.v \
  /another/thing/last \
  check_pp_final_state.svh
```

To execute special/custom preprocessor statements, you would put them into `/another/thing/first` and `/another/thing/last`.
@Sv3n What do you think?

DaveMcEwan, 2023-11-11:
@Sv3n If you think the above will meet your needs, you can try it on [this branch](https://github.com/DaveMcEwan/svlint/tree/feature269).

This command...
```sh
$ SVLINT_PREFILES='preA:preB' \
  SVLINT_POSTFILES='postA:postB' \
  SVLINT_INCDIRS='incA:incB' \
  cargo run -- --dump-filelist=yaml -IincX -IincY cliX.sv cliY.sv
```

...gives the output:
```
".":
  files:
    - "preA"
    - "preB"
    - "cliX.sv"
    - "cliY.sv"
    - "postA"
    - "postB"
  incdirs:
    - "incA"
    - "incB"
    - "incX"
    - "incY"
  defines:
```

Sv3n, 2023-11-12:
It makes sense to avoid the config file if you consider that it is shared in a team, good point. I think the environment variables approach should work nicely. I can give it a test drive tomorrow.

Sv3n, 2023-11-13:
@DaveMcEwan : Thanks for your efforts so far! I've tried the branch. The path-splitting goes wrong on Windows at the moment, because ':' has a different meaning there. After changing `e.split(':').map(|x| PathBuf::from(x)).collect()` into `env::split_paths(&e).map(|x| PathBuf::from(x)).collect()` and recompiling, it works, as long as SVLINT_PREFILES contains only absolute paths. Not sure if that last part is intentional.

For context, I'm testing this on Windows 11, and originally got here through:
- this VS Code extension: https://github.com/mshr-h/vscode-verilog-hdl-support
- which supports svls as language server
- which in turn uses svlint
(but I'm testing the new things using the plain svlint.exe)

DaveMcEwan, 2023-11-14:
Woops! Thanks for uncovering my quick hack :p I have updated my branch and opened #271. Thanks for the request and testing :)

Yes, it's intentional that the files listed in `SVLINT_PREFILES` (and `SVLINT_POSTFILES`) are not depending on `SVLINT_INCDIRS`. Include directories are only searched with preprocessor `include` directives (nothing on the command-line).

Sv3n, 2023-11-16:
I've tried the updated development branch, and it works as expected :)

DaveMcEwan, 2023-12-06:
@Sv3n If the release of v0.9.1 is satisfactory, would you mind closing this?

closed 2023-12-07 by Sv3n