dalance/svlint #273: New Rule: `procedural_continuous_assignment`

remes-codasip, 2023-11-21:
The following code contains "Procesural Continuous Assignment" statement  (`assign` in an `always` block) which is not synthesizable (at least not by all tools).

```systemverilog
module test (
    input logic clk,
    input logic a,
    input logic b,
    output logic c
);

always_ff @(posedge clk)
    assign c = a + b;

endmodule
```

Can you please add a rule to check for such constructs?

DaveMcEwan, 2023-12-06:
Seems like a sensible idea. From the IEEE1800-2017, page 242:
> NOTE - The procedural assign and deassign constructs are under consideration for deprecation. See Annex C.

@remes-codasip Would you like some assistance to create this rule? There are some guidelines [here](https://github.com/dalance/svlint/blob/master/CONTRIBUTING.md#adding-a-new-rule). I think you'd be looking to detect either [`ProceduralContinuousAssignmentAssign`](https://docs.rs/sv-parser/latest/sv_parser/struct.ProceduralContinuousAssignmentAssign.html) or other variants of [`procedural_continuous_assignment`](https://docs.rs/sv-parser/latest/sv_parser/enum.ProceduralContinuousAssignment.html), depending on exactly what you're looking for. To implement this, you could modify [this rule](https://github.com/dalance/svlint/blob/master/src/syntaxrules/module_nonansi_forbidden.rs).

remes-codasip, 2023-12-07:
@DaveMcEwan thanks for the hints you sent me.

I have created https://github.com/dalance/svlint/pull/276 and would welcome feedback (I think the documentation of the rule could be better, but I'm not familiar enough with SystemVerilog to write it better).

DaveMcEwan, 2024-04-01:
@remes-codasip @dalance The PR was merged so this can be closed.

closed 2024-04-01 by dalance