dalance/svlint #275: Feature request: ignoting lintnig (applying rules)

Mishassio, 2023-11-24:
Could you please add command lines option that would allow to ignore the rules for defined directoriese and includes.
For example:
svlint --dont-lint-includes filename. sv --dont-lint-incdir /foo_path/folder_with_src/
It would be usefull if we could use these options inside filelists.

This feature is needed for linting user filesets which include uvm sources with another code style.

DaveMcEwan, 2023-12-07:
This functionality is already in the TOML config, called `exclude_paths`, and is described [here](https://github.com/dalance/svlint/blob/master/md/manual-introduction.md#option-section).

I think it's safer to keep this in the config file so that it's easier to track (git, SVN, ...) what you're *not* checking - with CLI options, they're not tracked.

Does this help you?

DaveMcEwan, 2023-12-17:
@Mishassio Sorry man, I'm not sure what you're asking. Maybe you could make a draft PR that does what you need?

Mishassio, 2024-01-31:
We have solve problem with existing functionality

closed 2024-01-31 by Mishassio