dalance/svlint #282: Request: add a sensible default configuration to the README

DeflateAwning, 2024-03-23:
I'm at a loss of how to get started using this tool. There are so many conflicting linting rules that I don't know where to go. 

Is there a sensible default configuration that can be shared and linked in the README? It'd make this tool so much more helpful.

ronitnallagatla, 2024-03-29:
You can refer to the [Rulesets](https://github.com/dalance/svlint/blob/master/MANUAL.md#rulesets) section of the manual that describe various sets of pre-configured rules. Hope that helps!

DeflateAwning, 2024-03-29:
Why isn't there a sensible default set? The idea that the default is completely unusable seems insane to me

DaveMcEwan, 2024-04-01:
What's sensible for one use-case, e.g. a synthesizable design, isn't quite the same for another use-case, e.g. verification. There are rulesets precisely because there isn't a one-size-fits-all solution. As rotitnallagatla suggested, you can read the manual.

DeflateAwning, 2024-04-01:
I understand the desire to customize.

In regular languages (e.g., Rust, Python, etc.), embedded development vs. web development generally have different goals, difference frameworks, different editor configurations, etc.; but that doesn't stop the makers of Pyright and rust-analyzer from setting sensible defaults that allow users to get a feel for the tools before having to read the manual just to try writing a few modules of Verilog.

Why is the hardware design space so opposed to anything being usable-by-default? I'm not asking for perfect; I'm just asking for it to demo anything useful at all. 

Anyway, I've read the manual. I'm a beginner playing with a toy FPGA. **What configuration do you suggest I go with?**

DaveMcEwan, 2024-04-01:
SystemVerilog is different from Rust and Python which are both software-only languages. SystemVerilog is used for writing software (that's typically for verification by simulation), and specifying hardware circuits (cosmetically similar but practically quite different), and formal specification (different from both). In software, you can get away with learning by just hacking around, but in hardware that's a long and difficult learning path. You'll have a lot less frustration if you begin by reading the LRM and really understand the programming model (IEEE 1800-2017 is the newest standard). Certainly, the first 6 chapters (about 50 pages) are well-written easy reading that will give you most of the fundamental background.

For beginning with an FPGA, I'd recommend the ruleset called `designintent`. That means running commands with `svlint-designintent` instead of `svlint`. That ruleset is specifically designed to help beginners avoid the common pitfalls - if your code passes and does mostly what you want, then there's a very low chance of unpleasant surprises and a very high chance you really understand your FPGA configuration.

keesj, 2025-03-24:
If there is **no good default** the user should perhaps be forced to choose a configuration. After spending some time simply running svlint and fixing issues I came to the conclusion that it was not possible (for me) to create a simple hello world  test bench file that would pass the default rules