dalance/svlint #290: Feature request:  Extend trait to optionally supply fixes

ollie-etl, 2024-12-11:
First, svlint is great. Its the only lint enforcing tool I know which handles macros, and having it point out all of the inconsisencies is great. What would be better is if it could be run in a format mode, where each rule could rewrite the offending section (rustfmt, brittany, etc)

Having not yet dub into the codebase, I don't know how much of a change that would be. I guess as a minimum
  - trait needs extending to include `rewrite(..)`, which would need to be optional.
  - when parsing macros and includes, the AST would need to recall which sections were original, including line numbers, and be able to reconstruct exactly without any changes.

DaveMcEwan, 2024-12-12:
I think this feature request should go on sv-parser, which svlint is built upon.

Essentially, you would implement a method on each type of CST node to produce a canonical string representation, e.g. `to_string_canonical()`. So starting at the leaf nodes (e.g. `TimeLiteral`) and work your way up the tree to `SourceText` There are [about 200](https://docs.rs/sv-parser/latest/sv_parser/source_text/index.html) types of CST node, so it's not a small job.
Once all CST nodes have that method, your `rewrite` in svlint would call `to_string_canonical()` on the CST root node and put the resultant string somewhere.

I think this would be a nice addition, and it would pave the way for a similar effort - something like `to_serializable()` on each node. That would enable using serde to store the CST in JSON/YAML/whatever so that tools can process it easily (if not efficiently).

ollie-etl, 2024-12-14:
Ok, thanks for your insights. I agree that svlint might not be the best fit. I think the scope of a formatter is whitespace only. On reflection, I think that other formatting changes, like style, redundant brackets etc, would be a better match in svlint. That at least seems to be the split in rust between `rustfmt` and `clippy`

Still christmas is coming up. I might take a look over the holidays